dfki-ric/phobos #342: AttributeError: module 'bgl' has no attribute 'glBegin'

reinzler, 2024-01-23:
AttributeError: module 'bgl' has no attribute 'glBegin'

**To Reproduce**
[image]
[image]

Steps to reproduce the behavior:
1. Blender 3.6.8
2. Phobos 2.0.0
3. Terminal output during the process of export model to urdf
4. See error

AlpenAalAlex, 2024-02-07:
@reinzler Please take a look at my PR and see if this fixes your error

reinzler, 2024-02-09:
> @reinzler Please take a look at my PR and see if this fixes your error

Hi, thanks, now I have such error:
[20240209_16:19:01] ERROR No properly defined models to export
[image]

hwiedPro, 2024-02-13:
@reinzler Thanks for reporting the issue and testing. This issue seems to be resolved.

To help with your other problem, we could need some more information. Have you defined the root link by giving a name to your model?
Can you share your model or a screenshot? Please also consider the basic modelling walkthrough in the wiki to check if you have done everything that's necessary.

closed 2024-02-13 by hwiedPro

reinzler, 2024-04-25:
> @reinzler Thanks for reporting the issue and testing. This issue seems to be resolved.
>
> To help with your other problem, we could need some more information. Have you defined the root link by giving [...]

Hi, sorry for a long time reply, I can share my stl models, not a problem, will it be helpful?