dfki-ric/phobos #363: Bug in phobos/core/multiple.py

m0rsch1, 2024-05-31:
https://github.com/dfki-ric/phobos/blob/e359008a2302cb7307e6ce9e0dc7e80feb5b3c4f/phobos/core/multiple.py#L173

The `inputfile` can be `None` but is not checked before calling `os.path.abspath(inputfile)`

AlpenAalAlex, 2024-06-06:
Can you give me the python traceback to this error?

m0rsch1, 2024-06-07:
Well, i dont have a trace right now. But i think its not needed here: If i pass no `inputfile` to the `__init__` function of the Arrangement class, then it is set to `None` per default. In that case the line i have highlighted will fail, because `os.path.abspath(None)` is an invalid call.

m0rsch1, 2024-06-07:
So, either the `inputfile` argument is needed, then it should not be set to `None` per default or (which i think has been intended) the `self.inputfile` attribute has to be initialized to `None` as well and overwritten only if the argument `inputfile` is not None (there is an if statement for that already later in the `__init__` constructor).

hwiedPro, 2024-06-13:
Please check with pre_v2.1.0 ;)

m0rsch1, 2024-06-14:
After a quick look into the affected file i dont see this bug fixed there.

hwiedPro, 2024-06-20:
But now it is;)

m0rsch1, 2024-08-22:
Found another similar issue here:
```
--- a/phobos/core/multiple.py
+++ b/phobos/core/multiple.py
@@ -22,7 +22,7 @@ class Entity(Representation, SmurfBase):
         assert world is not None
         self.model = _singular(model)
         self.origin = _singular(origin) if origin is not None else representation.Pose()
-        self._file = os.path.normpath(os.path.join(os.path.dirname(world.inputfile), file)) if not os.path.isabs(file) else file
+        self._file = os.path.normpath(os.path.join(os.path.dirname(world.inputfile), file)) if file is not None and not os.path.isabs(file) else file
         if model is None and file is not None:
             if self._file.lower().rsplit(".", 1)[-1] in ["smurfs", "smurfa"]:
                 try:
```

m0rsch1, 2024-09-02:
Has been fixed, thank you :)

commit 624ccba2 (2025-03-25): Fix #363 - 2
closed 2024-09-02 by m0rsch1