dfki-ric/phobos #372: Blender freezes when trying to import Phobos

jeffreyjene, 2025-01-14:
Blender will freeze up when trying to enable Phobos after importing via zip file. This happens with ANY version of Blender that I've tried. (3.3, 3.6, 4.1).

No console message. I have to kill Blender to exit.

GreenZhu233, 2025-02-09:
Probably because you don't have Python dependencies installed.

AlpenAalAlex, 2025-02-12:
Which os are you working with? See if you can find a blender.crash.txt or another log file. Also installing the dependencies might take a while

jeffreyjene, 2025-02-12:
I'm using Windows 11. I haven't tried using Linux yet. Are the instructions to install dependencies the same for Windows?

AlpenAalAlex, 2025-02-24:
This note might help you: [https://github.com/dfki-ric/phobos/wiki/Installation#getting-phobos](https://github.com/dfki-ric/phobos/wiki/Installation#getting-phobos) 
Did you open the system console? [https://github.com/dfki-ric/phobos/wiki/First-Steps#log-settings](https://github.com/dfki-ric/phobos/wiki/First-Steps#log-settings)

jeffreyjene, 2025-02-24:
I'll try this out. Just a note, I posted a tutorial to using Phobos here: https://www.youtube.com/watch?v=yUpXi-3mNsM&t=8s

If everything is accurate in the video, perhaps I'll redo it with the new installation steps.

jeffreyjene, 2025-03-01:
This seems to be working now after installing the latest Microsoft Visual C++ Redistributable. Closing the issue.

closed 2025-03-01 by jeffreyjene