dfki-ric/phobos #374: 为什么我export model但是没有模型文件？

FSA9, 2025-02-11:
blender环境是3.6，插件版本2.0，导出模型之后没有模型文件，查了一下urdf文件里面也没有mesh的filename

AlpenAalAlex, 2025-02-12:
Hi, did you select a mesh file type to export?

[image]

Did you set the geometry type of your visual to mesh?

[image]

Please let me know if you figured it out

FSA9, 2025-02-13:
非常感谢，是我疏忽了，原先默认使用的是box，后来没意识到修改这个
不过现在似乎又遇到了一个问题，创建collision object的时候出现了报错：
AttributeError: The geometry of object center_collision has not yet been defined.
第一次还没遇到过这个问题，不知道为什么这次突然遇见了
还有Draw model information
这个始终都没有显示信息，这个是需要设置一些什么嘛

FSA9, 2025-02-13:
> Hi, did you select a mesh file type to export?
> 
> [image]
> 
> Did you set the geometry type of your visual to mesh?
> 
> [image]
> 
> Please let me know if you figured it out
非常感谢，是我疏忽了，原先默认使用的是box，后来没意识到修改这个
不过现在似乎又遇到了一个问题，创建collision object的时候出现了报错：
AttributeError: The geometry of object center_collision has not yet been defined.
第一次还没遇到过这个问题，不知道为什么这次突然遇见了
还有Draw model information
这个始终都没有显示信息，这个是需要设置一些什么嘛

FSA9, 2025-02-13:
> Hi, did you select a mesh file type to export?
> 
> [image]
> 
> Did you set the geometry type of your visual to mesh?
> 
> [image]
> 
> Please let me know if you figured it out

#
非常感谢，是我疏忽了，原先默认使用的是box，后来没意识到修改这个
不过现在似乎又遇到了一个问题，创建collision object的时候出现了报错：
AttributeError: The geometry of object center_collision has not yet been defined.
第一次还没遇到过这个问题，不知道为什么这次突然遇见了
还有Draw model information
这个始终都没有显示信息，这个是需要设置一些什么嘛

AlpenAalAlex, 2025-02-13:
Draw model information does not affect the export. For your center_collision you have to define the geometry like I did in my second picture. It has to be defined for every visual and collision.

FSA9, 2025-02-13:
> 绘制模型信息不会影响导出。对于你的center_collision你必须定义几何体，就像我在第二张图片中所做的那样。必须为每个视觉对象和碰撞定义它。

可是我已经定义了啊，

[image]

[image]

[image]
总共就两个部分，反倒是我在第一次尝试的时候却没有问题

FSA9, 2025-02-13:
[image]
这是我第一次尝试的时候制作的模型，这个时候一切正常没有任何问题，但是这次就不知道为什么不行了

closed 2025-02-14 by FSA9