dfki-ric/phobos #375: create collision object 以及create inertial的问题

FSA9, 2025-02-14:
在create collision object的时候，如果是第一次create的话，就会出现 The geometry of object center_collision has not yet been defined.的问题，因为默认的是box，但是当第一次从左下角窗口设置成mesh之后，就不会在报错了，测试后发现除了mesh以外用其他的类型都会报这个类型的错，可是我的原物体已经设置了visual和mesh
然后，当create collision object的时候，物体都是在世界原点生成的，并不是像视频中一样生成在选择物体中心
还有在create inertial的时候，有的时候会莫名报错，nonetype has no attribut input，然后后面随便点了几下又莫名其妙的好了？？？
这是我的操作步骤，可以麻烦看一下哪里有问题吗

FSA9, 2025-02-14:
[1.zip](https://github.com/user-attachments/files/18799906/1.zip)

AlpenAalAlex, 2025-02-24:
Sounds like you're using the outdated release. Please install this release: [phobos 2.0.1](https://github.com/dfki-ric/phobos/releases/tag/2.0.1)

FSA9, 2025-02-26:
> 听起来您使用的是过时的版本。请安装此版本： [phobos 2.0.1](https://github.com/dfki-ric/phobos/releases/tag/2.0.1) 请告诉我这是否解决了您的问题

似乎还是不行，错误跟之前视频里的一模一样，而且我下载了你这个版本在blender中显示依然是2.0.0

FSA9, 2025-02-26:
> Sounds like you're using the outdated release. Please install this release: [phobos 2.0.1](https://github.com/dfki-ric/phobos/releases/tag/2.0.1) Let me know if this fixes your issue
似乎还是不行，错误跟之前视频里的一模一样，而且我下载了你这个版本在blender中显示依然是2.0.0

zebra314, 2025-02-28:
I also met the same problem that the collision objects are generated from the world origin. I've tried to download the newest version from [here](https://github.com/dfki-ric/phobos/releases/tag/2.0.1), but couldn't solve it.

[image]

AlpenAalAlex, 2025-03-18:
@FSA9 @zebra314 Please let me know if my recent commit fixes your issues

closed 2025-04-16 by AlpenAalAlex